derpicknicker1/OctoPrint-Mmu2filamentselect #1: Not installed correctly

EddyMI3d, 2019-10-04:
Hi!
I just tested you marvellous idea with this plugin.
Sadly it does not install properly. 
After installing I get this message:
[image]
(Read: 
Unknown installed 
A restart of OctoPront is necessary...)

There is no entry in the plugin list, neither in the settings nor in the plugin manager.

The pluginmanager log gives no clue:
```
2019-10-04 19:28:33,270   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/derPicknicker1/OctoPrint-Mmu2filamentselect/archive/master.zip --no-cache-dir
2019-10-04 19:28:36,375 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2019-10-04 19:28:36,376 > Collecting https://github.com/derPicknicker1/OctoPrint-Mmu2filamentselect/archive/master.zip
2019-10-04 19:28:47,388 > Downloading https://github.com/derPicknicker1/OctoPrint-Mmu2filamentselect/archive/master.zip
2019-10-04 19:28:52,449 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Mmu2filamentselect==1.0.1) (1.3.12rc3)
2019-10-04 19:28:52,450 > Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.10.1)
2019-10-04 19:28:52,450 > Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (2.8.1)
2019-10-04 19:28:52,451 > Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (4.5.3)
2019-10-04 19:28:52,451 > Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.2.11)
2019-10-04 19:28:52,452 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (2019.1.24)
2019-10-04 19:28:52,452 > Requirement already satisfied: Flask-Principal<0.5,>=0.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.4.0)
2019-10-04 19:28:52,452 > Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.12.2)
2019-10-04 19:28:52,453 > Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.12)
2019-10-04 19:28:52,454 > Requirement already satisfied: werkzeug<0.16,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.15.2)
2019-10-04 19:28:52,454 > Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.1)
2019-10-04 19:28:52,455 > Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (5.1)
2019-10-04 19:28:52,455 > Requirement already satisfied: markdown<3.1,>=3.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (3.0.1)
2019-10-04 19:28:52,455 > Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (3.4)
2019-10-04 19:28:52,456 > Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.7.19)
2019-10-04 19:28:52,456 > Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.9.0)
2019-10-04 19:28:52,456 > Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.1.5.post0)
2019-10-04 19:28:52,457 > Requirement already satisfied: netifaces<0.11,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.10.9)
2019-10-04 19:28:52,457 > Requirement already satisfied: pylru<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.2.0)
2019-10-04 19:28:52,457 > Requirement already satisfied: rsa<5,>=4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (4.0)
2019-10-04 19:28:52,458 > Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.5.0.1)
2019-10-04 19:28:52,459 > Requirement already satisfied: requests<3,>=2.21.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (2.21.0)
2019-10-04 19:28:52,459 > Requirement already satisfied: semantic_version<2.7,>=2.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (2.6.0)
2019-10-04 19:28:52,460 > Requirement already satisfied: psutil<5.7,>=5.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (5.6.1)
2019-10-04 19:28:52,460 > Requirement already satisfied: Click<8,>=7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (7.0)
2019-10-04 19:28:52,460 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.6.5)
2019-10-04 19:28:52,461 > Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (5.2.1)
2019-10-04 19:28:52,461 > Requirement already satisfied: chainmap<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.0.3)
2019-10-04 19:28:52,462 > Requirement already satisfied: future<0.18,>=0.17.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.17.1)
2019-10-04 19:28:52,462 > Requirement already satisfied: scandir<1.11,>=1.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.10.0)
2019-10-04 19:28:52,462 > Requirement already satisfied: websocket-client<0.57,>=0.56 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.56.0)
2019-10-04 19:28:52,463 > Requirement already satisfied: wrapt<1.12,>=1.11.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.11.1)
2019-10-04 19:28:52,463 > Requirement already satisfied: futures<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (3.2.0)
2019-10-04 19:28:52,464 > Requirement already satisfied: emoji<0.6,>=0.5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.5.2)
2019-10-04 19:28:52,464 > Requirement already satisfied: monotonic<1.6,>=1.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.5)
2019-10-04 19:28:52,464 > Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.2)
2019-10-04 19:28:52,465 > Requirement already satisfied: sentry-sdk==0.7.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (0.7.7)
2019-10-04 19:28:52,465 > Requirement already satisfied: typing<4,>=3.6.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (3.6.6)
2019-10-04 19:28:52,465 > Requirement already satisfied: filetype<2,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.0.5)
2019-10-04 19:28:52,466 > Requirement already satisfied: itsdangerous>=0.21 in /home/pi/oprint/lib/python2.7/site-packages (from flask<0.11,>=0.10.1->OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (1.1.0)
[... cut ...]
2019-10-04 19:28:52,471 > Requirement already satisfied: pytz>=0a in /home/pi/oprint/lib/python2.7/site-packages (from Babel>=2.3->Flask-Babel<0.13,>=0.12->OctoPrint->OctoPrint-Mmu2filamentselect==1.0.1) (2018.9)
2019-10-04 19:28:52,472 > Installing collected packages: OctoPrint-Mmu2filamentselect
2019-10-04 19:28:52,472 > Running setup.py install for OctoPrint-Mmu2filamentselect: started
2019-10-04 19:28:55,477 > Running setup.py install for OctoPrint-Mmu2filamentselect: finished with status 'done'
2019-10-04 19:28:57,747 > Successfully installed OctoPrint-Mmu2filamentselect-1.0.1
```

ruedli, 2019-10-04:
Indeed, can confirm the same behavior, I am using an English version and it show the same dialog in English. After installing / restarting the plugin does not show up in the list of installed plugins: you can install it again, but there is no effect: a GCODE file with a Tx in it, still results in a prompt on the printer.

nullsibnelf, 2019-10-04:
same issue on mine 2 printers. both running pi Zero via pinheader on the einsy-board.
  running   OctoPrint 1.3.11 on OctoPi 0.16.0 

installed several times with reboots. cant see any change.

HenrikH96, 2019-10-05:
Would love to use this. But it's not installing correctly for me either.

wengeol2, 2019-10-05:
plug-in didn't install correctly for me either... installer sees it as "unknown". doesn't show in the Add-in screen and doesn't trigger. Probably a small metadata issue.. looking forward to using this plug-in so that I can remotely start a print in single mode

nullsibnelf, 2019-10-06:
Have to add, that since i tried to nstall the plugin, my other "printer-Alert"-Plugin desnt work anymore :-/

wengeol2, 2019-10-06:
I guess you should be able to remove it with
`pip uninstall mmu2filamentselect`
I will try that in about 5 hours when my printer is done printing a 46 hour monster print. not risking it before

RussPitcher, 2019-10-06:
Same here.  Installed it twice but nothing appears after the restart. Looking forward to a fixed version :)

TheFitzZZ, 2019-10-06:
Chiming in - same issue as the others :-)
Thank you @derpicknicker1 for building this neat addon!

derpicknicker1, 2019-10-06:
I'm very sorry. It's fixed now with version [1.0.2](https://github.com/derpicknicker1/OctoPrint-Mmu2filamentselect/releases/tag/1.0.2)
Thanks @all for reporting

closed 2019-10-06 by derpicknicker1

nullsibnelf, 2019-10-07:
Thanks a lot! Works great now! <3