derpicknicker1/OctoPrint-Mmu2filamentselect #3: TouchUI filament Select screen - Doesn't close

TheNore, 2019-10-06:
I must first thank you SO much for developing this plugin! I work remotely and needed this option for SO long!  The issue I am describing is minor. However I am just documenting it for future users / development. Everything works perfectly, but in the touch UI display, the selection window doesn't close when I select a Filament on the GUI web version of octoprint, and Vice versa. I just have to hit the same number on the screen. I am not sure if the Framework allows you to stop display on multiple instances of octoprint, but I just wanted to make sure. thank you for all of your work put into this.

derpicknicker1, 2019-10-06:
I'm not at home for a few days, but i will investigate on that as soon as possible.

TheNore, 2019-10-07:
great thank you!