derpicknicker1/OctoPrint-Mmu2filamentselect #4: Support "T?"

herm, 2019-10-08:
According to the [offical documentation](https://github.com/prusa3d/Prusa-Firmware/wiki/Supported-G-codes#t-codes) the code "T?" is used for selecting the extruder. This is also the code emitted by PrusaSlicer by default. However this code is ignored by this plugin. Could you please add support for it?

ruedli, 2019-10-10:
This was indeed the case for older profiles. The profiles I have now all use Tx / Tc. T? is still supported by the firmware though and in fact should be replaced by a T0..T4, whereas Tx should not. See also issue #2

derpicknicker1, 2019-11-01:
Since i do not own a MMU1, I have no chance to test the changes. @herm do you own a MMU1 to test the functionality if i implement the support for MMU1?

herm, 2019-11-01:
Yes, I will test it.