derpicknicker1/OctoPrint-Mmu2filamentselect #9: Suggestion request for Filament ID/color

wdgeek, 2019-11-04:
First want to let you know that I really appreciate the MMU2 selector plugin. Works GREAT. Had a suggestion to make. When the  select Filament page shows would it be possible to have  a user defined description of the filament showing along with "Filament 1". So  "Filament 1" would show up as "Filament 1 PETG White Overture" and "Filament 2 PETG Black Esun" etc. This would eliminate having to remember what is currently loaded into the MMU2's 5  filament positions (20 should be enough if I wanted to also have the manufacturer of the filament included in the description, but 15 would be ok without the mfg info.). Anyway,  thanks again for the great plugin.

derpicknicker1, 2019-11-05:
#5 asked for an integration of filament manager plugin. maybe it is possible that way.

smitty97, 2020-04-25:
I like this idea.  Could just have 5 input boxes on the plugin setup screen where you set timeout

tkoecker, 2020-09-11:
I was missing the same thing, see PR #23