derpicknicker1/OctoPrint-Mmu2filamentselect #13: MMU2S attempts to load filament while hotend is cold, grinds filament

akf-was-here, 2020-01-04:
Not sure is this is a repeat of issue #2 but just finished working with prusaslicer support (https://github.com/prusa3d/PrusaSlicer/issues/3448) figuring out that this plug-in is causing:

When starting a print with the printer field set to "Original Prusa i3 MK3S MMU2S Single" the gcode generated attempts to load the filament to the hotend before it's heated. I see the MMU2s send the filament down to the hotend and then jam up against the extruder gear causing the MMU2s gears to grind on the filament as it tries to push. Eventually it grinds enough out to lose purchase on the filament and everything is stuck until I clear it.  This only happens when starting from cold.

When I unscrew the bottom of the bowden from the extruder and issue the **Tx** it sends filament down to ~50mm past the end of the PTFE coupling and then slowly advances it by ~40mm at a time until it gives up at ~800mm, retracts ~1 bowden length, then starts stepping backwards ~40mm at a time until the mmu2 sensor detects a clear path and starts the process again. After the 2nd attempt it gives up and flags an "MMU Load Failed" on the LCD. I can halt the send process by manually actuating the extruder door to force an IR sensor trip. Note: these lengths are approximate because it moves too often for me to get a super accurate measurement and there is a lot to physically juggle but I think you can get the general picture.

Without this plugin (using the FW menu to select the filament) the MMU2s sends the filament down to ~40mm past the end of the bowden and once the hotend is up to temp (after the **M109** completes) the extruder gear spins and sends the filament on down to nozzle (I believe this is the **Tc** command).

ruedli, 2020-01-28:
This is indeed the behavior that I also found problematic in #2. Work-around is to ask the question later, so that the extruder is "hot', or substitute T? by an M701 E(n), which "only" loads till the FINDA and positions the selector. Then the TC should (when the user provided a selection on the octoprint prompt) that T<n>, as THEN the extruder is hot.  If you did not provide a choice, it should transmit TC. 

In the meantime I found, that with one extrac line of C++ code in the firmware you can support an optional command for specifying the extruder with T?. Then it will load EXACTLY like T? does, but you can specify the extruder to load with the T? instead of waiting for a prompt. T? E(0) would then behave in loading exactly as T?, be it with the SPECIFIED rather then interactively selected extruder.

Personnaly I like this best, but it requires a change in the Prusa firmware and sofar I was not very successful with my pull-requests towards Prusa, because Prusa only prioritizes pull requests that offer a benefit from the printer perspective: not for Octoprint plug-ins.

ruedli, 2020-03-02:
No: no new releases. I am waiting to proceed when I can configure the plugin to my needs. As it is I do not like its behavior. To be precise: I would like to be able to configure which GCODE commands are replaced by which alternative GCODE, as I explained in issue #2 to make the substitution of the plugin configurable.

sarf2k4, 2020-10-10:
I take it that this issues has not been solved yet. As for the time being, I use the following method to address the issues temporarily in gcode section of the setting. Reason being is that Prusa printers cold extrusion limit are set to 175c, setting to 180c is the safe temp to do load/unload sequence without grinding nor printer yelling with `COLD EXTRUSION PREVENTED`

**Before print starts:**
`M109 S180`

ruedli, 2020-10-10:
That works, however it meant I needed to update all existing gcodes. For
now the convenience to select from browser, did not outway reslicing with
updated settings. So... Until I can make the config settings needed myself
(to make it work without updated profiles), I deactivated the plugin. I
explained before how in my opinion should work to perform transparently.
Thanks for the update anyhow!

sarf2k4, 2020-10-10:
Yes, I understand that as well, the stock plugin parser(?) only replaces the Tx to Tn that we chose, but the printer has cold extrusion prevention and won't move the extruder motor until the current temperature exceeded 175c. I'm not familiar with the programming of the plugins. I also tried to disable prevent cold extrusion, but prusa firmware removed that feature.

ruedli, 2020-10-11:
indeed, upon Tx, the plugin should store the selection and just issue a
"load filament" (set the selector position and Not load filament further).
Then upon Tc the load into the nozzle should be executed, using the
selected toolpath.

The firmware currently does not support a command to load the filament to
the extrude (but not into the nozzle): much like Tx does. I looked into the
MK3S firmware and it is not hard to add as an option to Tx to actually
provide a selected extruder (as an optional argument. like E0 to E4) and
execute the load accordingly. give this: "just"  loading filament (so
placing the selector correctly) is the next best thing.

This would prevent cold-loads.

sarf2k4, 2020-10-11:
Does mk3s mmu2s supports M701 command? 

Furthermore, I think upon issuing a Tx command on mk3s mmu2s, the extruder motor will keep spinning until the sensor has been triggered, then will continue spin for a bit, in a sense of extrude (n)mm which I don't know how much as well. I don't think we can access the state of the sensors, finda and IR states by issuing a command like M119.

I wonder if there's any gcode commands that would do the following;; extrude x number of mm after filament sensor has been triggered. In a sense, this is similar to a stock MK3s direct extruder "auto filament load" feature, you stick in the filament, triggers the IR sensor and extruder will keep extrude.

I believe this kind of command would be useful if it's available in a form of gcode