derpicknicker1/OctoPrint-Mmu2filamentselect #19: Broken in FW 3.9.0-3421

mylife4aiurr, 2020-05-31:
The octoprint prompt works. When a mmu2 print is detected, a pop window asks for the filament choice. 
However choice is ignored, mmu2 uses what ever the selector is over at the time. The mk3's screen displays the F?, so no filament selection made it to mmu2.

timmmccoy, 2020-06-06:
I'm having the same issue, Just installed this plugin on a fresh Octoprint install.

wcraigtrader, 2020-07-07:
I'm having the same problem.I'm seeing this message in the Octoprint terminal:

> Not queuing T2, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware

Consequently, no T* message is sent, so the printer uses whatever filament was already selected.