derpicknicker1/OctoPrint-Mmu2filamentselect #21: [Feature Request] Python 3 Compatibility

chuckgenco, 2020-09-10:
Please add Python 3 Compatibility as outlined [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html).

wenca66, 2020-09-10:
Please me too "add Python 3 Compatibility". Thank you.

wenca66, 2020-09-11:
> See PR #22
> Till it is merged you can install from https://github.com/tkoecker/OctoPrint-Mmu2filamentselect/archive/development.zip

Thank you  - tested and currently works without problems!

smitty97, 2020-09-12:
> See PR #22
> Till it is merged you can install from https://github.com/tkoecker/OctoPrint-Mmu2filamentselect/archive/development.zip

does your dev version also have your filament labels? :)

tkoecker, 2020-09-12:
No, I wanted to keep those changes separate for the pull requests. But if you want to have both, I've merged them into:
https://github.com/tkoecker/OctoPrint-Mmu2filamentselect/archive/filament_labels_dev.zip

DaReaLDeviL, 2020-12-14:
The plugin still gets flagged as invalid in the Octoprint libary. I'm not that skilled, but when get it pushed there? Have some trouble with it actual and try so sort out where it comes from.

dajosco, 2021-02-07:
> The plugin still gets flagged as invalid in the Octoprint libary. I'm not that skilled, but when get it pushed there? Have some trouble with it actual and try so sort out where it comes from.

I just installed the latest version and works like a charm with labels and all! thanks!! 
[](https://github.com/tkoecker/OctoPrint-Mmu2filamentselect/archive/master.zip)

Would be awsome if it is added to the repository