derpicknicker1/OctoPrint-Mmu2filamentselect #28: Plugin is abandoned?

Patronics, 2021-01-20:
Given that there's 3 issues requesting Python 3 Compatibility (#17, #21, #25), and a pull request to fix it (#22), but no response to any of them from the Maintainer, (Also no response on the latest 15 issues on any subject and last 2 pull requests), it seems clear that this plugin is abandoned. The last activity from the maintainer I can find was a very succinct response of "No", to a question of whether it was abandoned nearly a year ago, in issue #14.

TheNore, 2021-01-29:
this is unfortunate, I heavily depend on this plugin, and now it is unusable on octopi .18
is there any way to get development started back ? I would gladly throw a donation to anyone willing to get this back up and going.

NightSkySK, 2021-02-01:
Do you know if @foosel has any established WoW with the abandoned plugins? Is it somehow possible to replace @derpicknicker1 version of plugin with branch/fork updated by @tkoecker which is prepared for Python 3?

foosel, 2021-02-01:
We have an abandoned plugin process in place for cases like this:

https://github.com/OctoPrint/plugins.octoprint.org/issues/new?assignees=&labels=%5B%22abandoned+plugin%22%5D&template=abandoned_plugin.md&title=%5BAbandoned+Plugin%5D+%3Cinsert+name+of+plugin+here%3E

ThePiGuy, 2021-02-22:
This plugin is now maintained at: https://github.com/tkoecker/OctoPrint-Mmu2filamentselect